TommyChangUMD/ros-humble-ros1-bridge-builder #2: tf subscribe failed

er1c-cartman, 2024-01-03:
Hi. first of all, thanks for sharing your great work
For me, it works well. I've made a 20.04 noetic docker on ubuntu 22.04 humble environment and succeed to run noetic package in this environment.
But the problem I'm facing now, the bridge isn't working on tf.
This error is keep occurring

failed to create 2to1 bridge for topic '/tf' with ROS 2 type 'tf2_msgs/msg/TFMessage' and ROS 1 type '': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option

So due to this error, I can run the simulation but couldn't communicate between ros1 and ros2 because there's no tf data.
Is there any way to bridge tf between ros1 and ros2?

Best regards

TommyChangUMD, 2024-01-03:
Hi er1c-cartman,

I have updated the Dockerfile to add support for tf2.   tf2_msgs should work now.  You can verify it by:

```
$ ros2 run ros1_bridge dynamic_bridge --print-pairs | grep -i tf
  - 'sensor_msgs/msg/NavSatFix' (ROS 2) <=> 'sensor_msgs/NavSatFix' (ROS 1)
  - 'sensor_msgs/msg/PointField' (ROS 2) <=> 'sensor_msgs/PointField' (ROS 1)
```

TommyChangUMD, 2024-01-03:
Hi er1c-cartman,

I have updated the Dockerfile to add support for tf2.  It should work now.  You can verify it by:

```
$ ros2 run ros1_bridge dynamic_bridge --print-pairs  | grep tf
  - 'tf2_msgs/msg/TF2Error' (ROS 2) <=> 'tf2_msgs/TF2Error' (ROS 1)
  - 'tf2_msgs/msg/TFMessage' (ROS 2) <=> 'tf2_msgs/TFMessage' (ROS 1)
  - 'tf2_msgs/srv/FrameGraph' (ROS 2) <=> 'tf2_msgs/FrameGraph' (ROS 1)

```

er1c-cartman, 2024-01-04:
thanks for the reply 
after I saw your comment, I've followed from the first, but the output is this

home@home:~$ ros2 run ros1_bridge dynamic_bridge --print-pairs  | grep tf
home@home:~$ 

should I have to delete the previous one and reinstall? 

If yes, how can I delete?

Sorry for asking such a simple questions.
And thanks again for kind reply

Best regards

TommyChangUMD, 2024-01-04:
After you git pull, just start from scratch and build the docker image again.  It will overwrite the old docker image.

TommyChangUMD, 2024-01-04:
And then, follow the same instructions to run the bridge example.  No need to explicitly delete anything.

er1c-cartman, 2024-01-04:
I've deleted the build file and worked.
thanks for great help :)
I'll ask you again if there's more question

home@home:~ros2 run ros1_bridge dynamic_bridge --print-pairs  | grep tftf
  - 'tf2_msgs/msg/TF2Error' (ROS 2) <=> 'tf2_msgs/TF2Error' (ROS 1)
  - 'tf2_msgs/msg/TFMessage' (ROS 2) <=> 'tf2_msgs/TFMessage' (ROS 1)
  - 'tf2_msgs/srv/FrameGraph' (ROS 2) <=> 'tf2_msgs/FrameGraph' (ROS 1)
home@home:~$ 

Best regards

er1c-cartman, 2024-01-04:
One more question. Sorry for asking you such a lot of questions :(
The thing I'm working on it is use ros1's mapping package for cost map and use on navigation in ros2.
I've run the ros1's package and tried to subscribe on ros2 rviz2. But for the topic, it was possible to subscribe, but it's not showing on rviz2. Is this problem of bridge or my setting?
[image]

Best regards

er1c-cartman, 2024-01-04:
for specific information, I'm uploading screen
this is ros2 rviz2's screen for topic
[image]

and this is ros1 rviz's screen for topic

[image]

For further information, I'm using elevation mapping for mapping.
but it isn't showing on ros2's topic

Best regards

er1c-cartman, 2024-01-04:
also some tf values are not subscribing

er1c-cartman, 2024-01-04:
sorry
everything works well I guess.
for the last part, can you add tf too? not tf2

failed to create 1to2 bridge for topic '/tf' with ROS 1 type 'tf/tfMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option

closed 2024-01-04 by er1c-cartman