TommyChangUMD/ros-humble-ros1-bridge-builder #3: failed to create 1to2 bridge for topic '/tf' with ROS 1 type 'tf/tfMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage': No template specialization for the pair

er1c-cartman, 2024-01-04:
sorry for asking you again...
everything works well I guess.
for the last part, can you add tf too? not tf2
i mean here's the error I'm facing. I tried to subscribe tf from ros1 to ros2 on rviz, but this error is occurring

failed to create 1to2 bridge for topic '/tf' with ROS 1 type 'tf/tfMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage': No template specialization for the pair
check the list of supported pairs with the --print-pairs option

Best regards

er1c-cartman, 2024-01-04:
specially it's not subscribing odom related topics. because some tf values such as base_link, imu_link etc is subscribing, but like odom, camera_odom things aren't subscribable.
Hope it will be solved. 
For more specific description, I'm trying to subscribe ros1's topic to ros2. Trying to show on rviz on ros2 of ros1's topic

Best regards.

TommyChangUMD, 2024-01-04:
Hi, Try running with ros1_bridge with the "--bridge-all-topics" option.

`$ ros2 run ros1_bridge dynamic_bridge --bridge-all-topics
`

er1c-cartman, 2024-01-04:
Hi.
yes I'm running with    
ros2 run ros1_bridge dynamic_bridge --bridge-all-topics

i think from ros2 to ros1, it works well, but from ros1 to ros2, it doesn't work specially with odom related 

I've checked with husky playpen and while running husky on ros1, ros2 couldn't subscribe odom tf
specially when I run rviz2, suddenly the odom's tf is gone
thanks for kind answering :)

Best regards.

er1c-cartman, 2024-01-04:
well I'll upload tomorrow what it looks like in bridge terminal 
Best regards :)

TommyChangUMD, 2024-01-04:
hmm.. I am not sure what might be the issue.   If it helps, the following pair shows how the odom messages are bridged:

```
$ ros2 run ros1_bridge dynamic_bridge --print-pairs | grep Od
  - 'nav_msgs/msg/Odometry' (ROS 2) <=> 'nav_msgs/Odometry' (ROS 1)

```

er1c-cartman, 2024-01-05:
Yes it works like this.
home@home:~$ ros2 run ros1_bridge dynamic_bridge --print-pairs | grep Od
  - 'nav_msgs/msg/Odometry' (ROS 2) <=> 'nav_msgs/Odometry' (ROS 1)

But the problem I'm having right now is this.
When I run husky package on ros1 terminal and after run rosbridge, this is what I can see on terminal

home@home:~$ ros2 run ros1_bridge dynamic_bridge --bridge-all-topics 
created 1to2 bridge for topic '/camera/fisheye1/camera_info' with ROS 1 type 'sensor_msgs/CameraInfo' and ROS 2 type 'sensor_msgs/msg/CameraInfo'
created 1to2 bridge for topic '/camera/fisheye1/image_raw' with ROS 1 type 'sensor_msgs/Image' and ROS 2 type 'sensor_msgs/msg/Image'
created 1to2 bridge for topic '/camera/fisheye2/camera_info' with ROS 1 type 'sensor_msgs/CameraInfo' and ROS 2 type 'sensor_msgs/msg/CameraInfo'
created 1to2 bridge for topic '/camera/fisheye2/image_raw' with ROS 1 type 'sensor_msgs/Image' and ROS 2 type 'sensor_msgs/msg/Image'
created 1to2 bridge for topic '/camera/gyro/sample' with ROS 1 type 'sensor_msgs/Imu' and ROS 2 type 'sensor_msgs/msg/Imu'
created 1to2 bridge for topic '/camera/odom/sample' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/clock' with ROS 1 type 'rosgraph_msgs/Clock' and ROS 2 type 'rosgraph_msgs/msg/Clock'
created 1to2 bridge for topic '/diagnostics' with ROS 1 type 'diagnostic_msgs/DiagnosticArray' and ROS 2 type 'diagnostic_msgs/msg/DiagnosticArray'
created 1to2 bridge for topic '/husky_velocity_controller/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
created 1to2 bridge for topic '/husky_velocity_controller/odom' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/imu/data' with ROS 1 type 'sensor_msgs/Imu' and ROS 2 type 'sensor_msgs/msg/Imu'
created 1to2 bridge for topic '/imu/data/bias' with ROS 1 type 'sensor_msgs/Imu' and ROS 2 type 'sensor_msgs/msg/Imu'
created 1to2 bridge for topic '/joint_states' with ROS 1 type 'sensor_msgs/JointState' and ROS 2 type 'sensor_msgs/msg/JointState'
created 1to2 bridge for topic '/kb_teleop/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
created 1to2 bridge for topic '/navsat/fix' with ROS 1 type 'sensor_msgs/NavSatFix' and ROS 2 type 'sensor_msgs/msg/NavSatFix'
created 1to2 bridge for topic '/navsat/vel' with ROS 1 type 'geometry_msgs/Vector3Stamped' and ROS 2 type 'geometry_msgs/msg/Vector3Stamped'
created 1to2 bridge for topic '/odom' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/odometry/filtered' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/points' with ROS 1 type 'sensor_msgs/PointCloud2' and ROS 2 type 'sensor_msgs/msg/PointCloud2'
created 1to2 bridge for topic '/rosout' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
created 1to2 bridge for topic '/rosout_agg' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
created 1to2 bridge for topic '/tf_static' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'
created 1to2 bridge for topic '/twist_marker_server/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
created 1to2 bridge for topic '/twist_marker_server/update' with ROS 1 type 'visualization_msgs/InteractiveMarkerUpdate' and ROS 2 type 'visualization_msgs/msg/InteractiveMarkerUpdate'
created 1to2 bridge for topic '/twist_marker_server/update_full' with ROS 1 type 'visualization_msgs/InteractiveMarkerInit' and ROS 2 type 'visualization_msgs/msg/InteractiveMarkerInit'
Created 2 to 1 bridge for service /camera/fisheye1/set_camera_info
Created 2 to 1 bridge for service /camera/fisheye2/set_camera_info
Created 2 to 1 bridge for service /ekf_localization/enable
Created 2 to 1 bridge for service /gazebo/pause_physics
Created 2 to 1 bridge for service /gazebo/reset_simulation
Created 2 to 1 bridge for service /gazebo/reset_world
Created 2 to 1 bridge for service /gazebo/unpause_physics
Created 2 to 1 bridge for service /imu/data/calibrate
[INFO] [1704416365.312643167] [ros_bridge]: Passing message from ROS 1 rosgraph_msgs/Clock to ROS 2 rosgraph_msgs/msg/Clock (showing msg only once per type)
[INFO] [1704416365.478210368] [ros_bridge]: Passing message from ROS 1 nav_msgs/Odometry to ROS 2 nav_msgs/msg/Odometry (showing msg only once per type)
[INFO] [1704416365.489626448] [ros_bridge]: Passing message from ROS 1 sensor_msgs/Imu to ROS 2 sensor_msgs/msg/Imu (showing msg only once per type)
[INFO] [1704416365.515987109] [ros_bridge]: Passing message from ROS 1 diagnostic_msgs/DiagnosticArray to ROS 2 diagnostic_msgs/msg/DiagnosticArray (showing msg only once per type)
[INFO] [1704416365.576237738] [ros_bridge]: Passing message from ROS 1 rosgraph_msgs/Log to ROS 2 rcl_interfaces/msg/Log (showing msg only once per type)
[INFO] [1704416365.576634801] [ros_bridge]: Passing message from ROS 1 tf2_msgs/TFMessage to ROS 2 tf2_msgs/msg/TFMessage (showing msg only once per type)
[INFO] [1704416365.577241164] [ros_bridge]: Passing message from ROS 1 visualization_msgs/InteractiveMarkerInit to ROS 2 visualization_msgs/msg/InteractiveMarkerInit (showing msg only once per type)
[... cut ...]
Also in bridge terminal, still showing this error
failed to create 1to2 bridge for topic '/tf' with ROS 1 type 'tf/tfMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option

So i think there's a problem with porting tf from ros1 to ros2.

If you can, maybe you can try to run husky_playpen package in ros1 and check rviz2 on ros2. You will see the same problem with me i guess

Thanks for helping me solve the issue

Best regards.

er1c-cartman, 2024-01-05:
Oh. and when I see the bridge terminal, it's only showing this
2 things are related with tf and it's these

created 1to2 bridge for topic '/tf_static' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'
[INFO] [1704418428.011715462] [ros_bridge]: Passing message from ROS 1 tf2_msgs/TFMessage to ROS 2 tf2_msgs/msg/TFMessage (showing msg only once per type)

So I guess the bridge is not porting tf...

And if it's porting, it's porting only once

er1c-cartman, 2024-01-05:
And sometimes it works but when I turn on the rviz2, it stops
bridge terminal

created 1to2 bridge for topic '/tf' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'
created 1to2 bridge for topic '/tf_static' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'

bridge terminal after I run rviz2 or check ros2 topic echo /tf

failed to create 1to2 bridge for topic '/tf' with ROS 1 type 'tf/tfMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage': No template specialization for the pair
check the list of supported pairs with the `--print-pairs` option

Best regards

TommyChangUMD, 2024-01-05:
I see.  Looks like the bridge know about

 **'tf2_msgs/TFMessage'** and ROS 2 type 'tf2_msgs/msg/TFMessage'

But not

**'tf/tfMessage'** and ROS 2 type 'tf2_msgs/msg/TFMessage':

Let me see if I can find the source code for "tf/tfMessage" and add it to Dockefile.

er1c-cartman, 2024-01-05:
Thanks for the quick and caring answer :)
Wish you the best

TommyChangUMD, 2024-01-05:
OK.  I updated the Dockerfile.   I should work now.

```
$ ros2 run ros1_bridge dynamic_bridge --print-pairs | grep tf
  - 'tf2_msgs/msg/TF2Error' (ROS 2) <=> 'tf2_msgs/TF2Error' (ROS 1)
  - 'tf2_msgs/msg/TFMessage' (ROS 2) <=> 'tf2_msgs/TFMessage' (ROS 1)
  - 'tf2_msgs/msg/TFMessage' (ROS 2) <=> 'tf/tfMessage' (ROS 1)
  - 'tf2_msgs/srv/FrameGraph' (ROS 2) <=> 'tf2_msgs/FrameGraph' (ROS 1)

```

er1c-cartman, 2024-01-05:
Thanks for updating. But unfortunately, it's still not working :(
The bridge terminal is not showing errors anymore but still can't find the odom frame in ros2 :(

This is ros2 rviz2's screen

[image]

And this is ros1 rviz's screen

[image]

I guess the odom related thing isn't working that well... specially with tf
Anyway thanks for spending your time to solve the issue.

Best regards :)

er1c-cartman, 2024-01-05:
And this is bridge terminal.
For me it isn't working well but your addition on tf is working

home@home:~$ ros2 run ros1_bridge dynamic_bridge --bridge-all-topics 
created 1to2 bridge for topic '/camera/fisheye1/camera_info' with ROS 1 type 'sensor_msgs/CameraInfo' and ROS 2 type 'sensor_msgs/msg/CameraInfo'
created 1to2 bridge for topic '/camera/fisheye1/image_raw' with ROS 1 type 'sensor_msgs/Image' and ROS 2 type 'sensor_msgs/msg/Image'
created 1to2 bridge for topic '/camera/fisheye2/camera_info' with ROS 1 type 'sensor_msgs/CameraInfo' and ROS 2 type 'sensor_msgs/msg/CameraInfo'
created 1to2 bridge for topic '/camera/fisheye2/image_raw' with ROS 1 type 'sensor_msgs/Image' and ROS 2 type 'sensor_msgs/msg/Image'
created 1to2 bridge for topic '/camera/gyro/sample' with ROS 1 type 'sensor_msgs/Imu' and ROS 2 type 'sensor_msgs/msg/Imu'
created 1to2 bridge for topic '/camera/odom/sample' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/clicked_point' with ROS 1 type 'geometry_msgs/PointStamped' and ROS 2 type 'geometry_msgs/msg/PointStamped'
created 1to2 bridge for topic '/clock' with ROS 1 type 'rosgraph_msgs/Clock' and ROS 2 type 'rosgraph_msgs/msg/Clock'
created 1to2 bridge for topic '/diagnostics' with ROS 1 type 'diagnostic_msgs/DiagnosticArray' and ROS 2 type 'diagnostic_msgs/msg/DiagnosticArray'
created 1to2 bridge for topic '/husky_velocity_controller/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
created 1to2 bridge for topic '/husky_velocity_controller/odom' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/imu/data' with ROS 1 type 'sensor_msgs/Imu' and ROS 2 type 'sensor_msgs/msg/Imu'
created 1to2 bridge for topic '/imu/data/bias' with ROS 1 type 'sensor_msgs/Imu' and ROS 2 type 'sensor_msgs/msg/Imu'
created 1to2 bridge for topic '/initialpose' with ROS 1 type 'geometry_msgs/PoseWithCovarianceStamped' and ROS 2 type 'geometry_msgs/msg/PoseWithCovarianceStamped'
created 1to2 bridge for topic '/joint_states' with ROS 1 type 'sensor_msgs/JointState' and ROS 2 type 'sensor_msgs/msg/JointState'
created 1to2 bridge for topic '/kb_teleop/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
created 1to2 bridge for topic '/move_base_simple/goal' with ROS 1 type 'geometry_msgs/PoseStamped' and ROS 2 type 'geometry_msgs/msg/PoseStamped'
created 1to2 bridge for topic '/navsat/fix' with ROS 1 type 'sensor_msgs/NavSatFix' and ROS 2 type 'sensor_msgs/msg/NavSatFix'
created 1to2 bridge for topic '/navsat/vel' with ROS 1 type 'geometry_msgs/Vector3Stamped' and ROS 2 type 'geometry_msgs/msg/Vector3Stamped'
created 1to2 bridge for topic '/odom' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/odometry/filtered' with ROS 1 type 'nav_msgs/Odometry' and ROS 2 type 'nav_msgs/msg/Odometry'
created 1to2 bridge for topic '/points' with ROS 1 type 'sensor_msgs/PointCloud2' and ROS 2 type 'sensor_msgs/msg/PointCloud2'
created 1to2 bridge for topic '/rosout' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
created 1to2 bridge for topic '/rosout_agg' with ROS 1 type 'rosgraph_msgs/Log' and ROS 2 type 'rcl_interfaces/msg/Log'
created 1to2 bridge for topic '/tf' with ROS 1 type 'tf/tfMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'
created 1to2 bridge for topic '/tf_static' with ROS 1 type 'tf2_msgs/TFMessage' and ROS 2 type 'tf2_msgs/msg/TFMessage'
created 1to2 bridge for topic '/twist_marker_server/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
created 1to2 bridge for topic '/twist_marker_server/feedback' with ROS 1 type 'visualization_msgs/InteractiveMarkerFeedback' and ROS 2 type 'visualization_msgs/msg/InteractiveMarkerFeedback'
created 1to2 bridge for topic '/twist_marker_server/update' with ROS 1 type 'visualization_msgs/InteractiveMarkerUpdate' and ROS 2 type 'visualization_msgs/msg/InteractiveMarkerUpdate'
created 1to2 bridge for topic '/twist_marker_server/update_full' with ROS 1 type 'visualization_msgs/InteractiveMarkerInit' and ROS 2 type 'visualization_msgs/msg/InteractiveMarkerInit'
Created 2 to 1 bridge for service /camera/fisheye1/set_camera_info
Created 2 to 1 bridge for service /camera/fisheye2/set_camera_info
Created 2 to 1 bridge for service /ekf_localization/enable
Created 2 to 1 bridge for service /gazebo/pause_physics
Created 2 to 1 bridge for service /gazebo/reset_simulation
Created 2 to 1 bridge for service /gazebo/reset_world
Created 2 to 1 bridge for service /gazebo/unpause_physics
Created 2 to 1 bridge for service /imu/data/calibrate
Created 2 to 1 bridge for service /rviz/reload_shaders
[INFO] [1704437351.885713793] [ros_bridge]: Passing message from ROS 1 rosgraph_msgs/Clock to ROS 2 rosgraph_msgs/msg/Clock (showing msg only once per type)
[INFO] [1704437351.957468598] [ros_bridge]: Passing message from ROS 1 sensor_msgs/Image to ROS 2 sensor_msgs/msg/Image (showing msg only once per type)
[INFO] [1704437351.962906758] [ros_bridge]: Passing message from ROS 1 nav_msgs/Odometry to ROS 2 nav_msgs/msg/Odometry (showing msg only once per type)
[INFO] [1704437351.962992491] [ros_bridge]: Passing message from ROS 1 sensor_msgs/Imu to ROS 2 sensor_msgs/msg/Imu (showing msg only once per type)
[... cut ...]
created 2to1 bridge for topic '/initialpose' with ROS 2 type 'geometry_msgs/msg/PoseWithCovarianceStamped' and ROS 1 type 'geometry_msgs/PoseWithCovarianceStamped'
created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
[INFO] [1704437352.971774610] [ros_bridge]: Passing message from ROS 1 sensor_msgs/NavSatFix to ROS 2 sensor_msgs/msg/NavSatFix (showing msg only once per type)
[INFO] [1704437352.972270444] [ros_bridge]: Passing message from ROS 1 geometry_msgs/Vector3Stamped to ROS 2 geometry_msgs/msg/Vector3Stamped (showing msg only once per type)

Best regards :)

er1c-cartman, 2024-01-05:
Or maybe it's not working well. I mean the bridge is working but it seems it's not getting the actual information.
I've checked with ros2 topic echo /tf, and the output is this

home@home:~$ ros2 topic echo /tf
transforms: []
---
transforms: []
---
transforms: []
---
transforms: []
---
transforms: []
---
transforms: []
---
transforms: []
---
transforms: []
---
transforms: []
---
transforms: []
---

I'm not sure but it seems the tf isn't ported well.

Thanks for your work on my issues

Best regards :)

TommyChangUMD, 2024-01-05:
Not sure if it helps, but looks like the same issue is reported here:
https://github.com/ros2/ros1_bridge/issues/219

TommyChangUMD, 2024-01-05:
Do you have to use **camera_odom** TF when running ros2 rviz2?

Just a hunch, how about changing the **Global Options->Fixed Frame** to use **camera_odom_frame** instead of **camera_odom** ?

> Thanks for updating. But unfortunately, it's still not working :( The bridge terminal is not showing errors anymore but still can't find the odom frame in ros2 :(
>
> This is ros2 rviz2's screen [...]

er1c-cartman, 2024-01-05:
well unfortunately, I have to use camera_odom. It's kind similar with odom, but using VO for using elevation mapping with traversability estimation. 
anyway camera_odom and odom tf isn't showing on rviz2 :(

sorry for not helping with your effort...

er1c-cartman, 2024-01-05:
I see. Thanks for spending your time for helping me :)
i really appreciate it
I'll try bit differently to solve the issue
Again, thanks for helping me to solve the issue
Wish you a good luck with your research

Best regards :)

TommyChangUMD, 2024-01-23:
Hi er1c-cartman, 

Can you try the latest Dockerfile? It might just work this time since I switched to using the pre-built binaries for tf and tf2.

er1c-cartman, 2024-01-24:
well I've tried it but doesn't work.
same problem i think. like cannot bridge odom related tf
but thanks for spending your time 
and i decided to solve the problem with just do migration by myself on ros2 environment
Best regards :)

closed 2024-01-24 by er1c-cartman