TommyChangUMD/ros-humble-ros1-bridge-builder #4: Action support

LucasLabarussiat, 2024-01-11:
Hello,

First of all thanks for your work, it helped a lot in understanding how this works, and I have been looking around quite some time to find a ros1_bridge between Noetic and Humble.
By any chance, have you tried to add support for actions ? I did try fiddling around with your base and the work from ipa-hsd (here are [his action_bridge branch](https://github.com/ipa-hsd/ros1_bridge/tree/action_bridge) and [the original issue on ros1_bridge](https://github.com/ros2/ros1_bridge/pull/256) ) but couldn't get it to work, still tyring to figure out why.

By the way, is there a reason to spit a tarball ? Would something like that work: 
- a first docker with ROS1 Noetic
- a second docker with the bridge and a ROS2 workspace ?
I guess you have some weird things done in the builder, so it might be better to get the built bridge and a ROS2 workspace on a fresh one.

Cheers,
Lucas

TommyChangUMD, 2024-01-12:
Hi Lucas,

You are correct that the builder does something to the apt database.  It's a workaround for the catkin issue.  You can read more about this issue here:  https://docs.ros.org/en/iron/How-To-Guides/Using-ros1_bridge-Jammy-upstream.html#ros-2-via-debian-packages

As for the ros1_bridge fork from ipa-hsd, I also ran into compilation issues:
`/ros-humble-ros1-bridge/src/ros1_bridge/include/ros1_bridge/action_factory.hpp:148:44: error: no match for ‘operator=’ (operand types are ‘rclcpp_action::Client<tf2_msgs::action::LookupTransform>::GoalResponseCallback’ {aka ‘std::function<void(std::shared_ptr<rclcpp_action::ClientGoalHandle<tf2_msgs::action::LookupTransform> >)>’} and ‘ros1_bridge::ActionFactory_1_2<tf2_msgs::LookupTransformAction_<std::allocator<void> >, tf2_msgs::action::LookupTransform>::GoalHandler::handle()::<lambda(std::shared_future<std::shared_ptr<rclcpp_action::ClientGoalHandle<tf2_msgs::action::LookupTransform> > >)>’)
`

I will try to investigate these compilation errors..

LucasLabarussiat, 2024-01-12:
Hi Thomas,

Thank you for your quick answer. I will also try to investigate, though I am not sure I am fully capable when it comes to this. If I find something I will let you know.

smith-doug, 2024-01-31:
The signature goal_response_callback wants changed to (const ROS2GoalHandle & gh2).  https://github.com/smith-doug/ros1_bridge/commit/1611633d47fdc33b299dbd7e066f3642812c7a4d#

TommyChangUMD, 2024-02-01:
Thanks, Douglas!  
I have updated the Dockerfile to use your fork of `action_bridge` (action_bridge_humble branch).

Lucas - could you try the latest Dockefile and see if actions support works for you?

LucasLabarussiat, 2024-02-05:
Hi,

Thanks for the help ! The bridge did build properly with your Dockerfile, though I haven't tried it and jumped right into trying to add my own dependencies.
Especially this package here: [industrial_robot_client](https://github.com/ros-industrial/industrial_core/tree/melodic-devel)

This is why I needed the bridge in the first place: I have an action server (control_msgs/msg/FollowJointTrajectory) running for my robot with an industrial_robot_client and want to bridge to an action client with ROS2.

I added `ros-humble-control-msgs`. If I get it right, I need to add git clone instructions for the ROS1 package and its dependencies and successfully build them using colcon. Working on this at the moment, hope I can get all dependencies. Using the command :
```
rosdep resolve `grep -RhI -e "^ *<[a-z,A-Z,_]*depend>" ~/ros_ws/src/industrial_core | sed -e "s/.*>\(.*\)<.*/\1/" | sort -r | uniq | tr "\n" " "` 2>/dev/null | sed -e "s/^#.*//" | tr "\n" " " | sed -e "s/  */ /g" -e "s/  *//" |  (echo -n "sudo apt-get install " && cat); echo ""
```
I got the following package dependencies :
```
 ros-noetic-urdf
ros-noetic-trajectory-msgs
ros-noetic-std-msgs
ros-noetic-simple-message
ros-noetic-sensor-msgs
ros-noetic-rosunit
ros-noetic-rospy
ros-noetic-roslaunch
ros-noetic-roscpp
ros-noetic-robot-state-publisher
ros-noetic-pluginlib
ros-noetic-moveit-ros-planning
ros-noetic-moveit-core
ros-noetic-message-runtime
ros-noetic-message-generation
ros-noetic-industrial-utils
ros-noetic-industrial-trajectory-filters
ros-noetic-industrial-robot-simulator
ros-noetic-industrial-robot-client
ros-noetic-industrial-msgs
ros-noetic-industrial-deprecated
ros-noetic-control-msgs
ros-noetic-class-loader
ros-noetic-catkin
ros-noetic-actionlib-msgs
ros-noetic-actionlib
```

If I can build the dependencies and then source the resulting `setup.bash` before building the bridge again, I should be able to use:
```
ros2 run ros1_bridge action_bridge ros1 industrial_robot_client FollowJointTrajectory joint_trajectory_action
```
Am I right ?

Thanks again,
Lucas

smith-doug, 2024-02-05:
I have run [Motoman's ros1 industrial robot client based driver](https://github.com/ros-industrial/motoman) from Humble with the bridge in Noetic.  I used [ros noetic in Conda](https://anaconda.org/ros-noetic-py10) that was build for python 3.10.  You'll need [this mapping package in your ros2 workspace](https://github.com/smith-doug/bridge_mapping).  There is a version of the [ros industrial core repo](https://github.com/ros-industrial/industrial_core/tree/ros2_msgs_only) that will give you the messages in ros2.
 
My fork is not in a finished state so use it at your own risk.  It did run well enough to run the[ ros industrial scan n plan workshop demo](https://github.com/ros-industrial-consortium/scan_n_plan_workshop/tree/master/snp_automate_2022).

I have not tried this Docker version.  I made a separate package for mappings before discovering the incompatibilities with noetic/humble messages are more than just mappings.  It might work with the mapping package in the ros2 /ros-humble-ros1-bridge/src folder.

LucasLabarussiat, 2024-03-27:
Hi, sorry for the delay.
Just wanted to let you know that everything works great now. My setup is as follows:
- a first container using your work to build the bridge
- a second one with ROS1 Noetic to run the staubli val3 driver and the action server
- a third one with ROS2 Humble to run the bridge and the action client

I want to contribute back somehow, but I first have to do some paperwork to have my company's approval.

Thanks a lot for your help.

TommyChangUMD, 2024-03-27:
Thanks for the feedback!  And feel free to contribute and make improvements.  :)

closed 2024-03-27 by TommyChangUMD