TommyChangUMD/ros-humble-ros1-bridge-builder #5: Failed to build on jetson

ambrosekwok, 2024-01-22:
Hello,

I tried to build the docker image in jetson nano with is arm64 cpu. Since the base image `osrf/ros:humble-desktop` is not available for arm64, I changed to `ros:humble` and removed the `ros_tutorials` packages.
I can build it in my amd64 cpu device but failed in jetson nano. Do you have any ideas to fix it? 

Thank you

Here is the modified dockerfile.
```
FROM ros:humble
#
# How to build this docker image:
#  docker build . -t ros-humble-ros1-bridge-builder
#
# How to build ros-humble-ros1-bridge:
#  # 0.) Start from the ROS 2 Humble system, build a "ros-humble-ros1-bridge/" ROS2 package:
#  docker run --rm ros-humble-ros1-bridge-builder | tar xvzf -
#
# How to use ros-humble-ros1-bridge:
#  # 1.) First start a ROS1 Noetic docker and bring up a GUI terminal, something like:
#  rocker --x11 --user --home --privileged \
#         --volume /dev/shm /dev/shm --network=host -- osrf/ros:noetic-desktop \
#         'bash -c "sudo apt update; sudo apt install -y tilix; tilix"'
#
#  # 2.) Then, start "roscore" inside the ROS1 docker
#  source /opt/ros/noetic/setup.bash
#  roscore
#
#  # 3.) Now, from the ROS2 Humble system, start the ros1 bridge node.
#  source /opt/ros/humble/setup.bash
#  source ros-humble-ros1-bridge/install/local_setup.bash
#  ros2 run ros1_bridge dynamic_bridge
#
#  # 3.) Back to the ROS1 Noetic docker container, run in another terminal tab:
#  source /opt/ros/noetic/setup.bash
#  rosrun rospy_tutorials talker
#
#  # 4.) Finally, from the ROS2 Humble system:
#  source /opt/ros/humble/setup.bash
#  ros2 run demo_nodes_cpp listener
#

# Make sure we are using bash and catching errors
SHELL ["/bin/bash", "-o", "pipefail", "-o", "errexit", "-c"]

# 0.) Bring system up to the latest
RUN apt -y update
RUN apt -y upgrade

# 1.) Temporarily remove ROS2 apt repository
RUN mv /etc/apt/sources.list.d/ros2-latest.list /root/
RUN apt update

# 2.) comment out the catkin conflict
RUN sed  -i -e 's|^Conflicts: catkin|#Conflicts: catkin|' /var/lib/dpkg/status
RUN apt install -f

# 3.) force install these packages
RUN apt download python3-catkin-pkg
RUN apt download python3-rospkg
RUN apt download python3-rosdistro
RUN dpkg --force-overwrite -i python3-catkin-pkg*.deb
RUN dpkg --force-overwrite -i python3-rospkg*.deb
RUN dpkg --force-overwrite -i python3-rosdistro*.deb
RUN apt install -f

# 4.) Install ROS1 stuff
# see https://packages.ubuntu.com/jammy/ros-core-dev
RUN apt -y install ros-core-dev

[... 140 lines cut ...]
  find_ros1_package() failed to find 'tf' using pkg_check_modules()
Call Stack (most recent call first):
  CMakeLists.txt:60 (find_ros1_package)

-- Configuring incomplete, errors occurred!
See also "/ros-humble-ros1-bridge/build/ros1_bridge/CMakeFiles/CMakeOutput.log".
--- stderr: ros1_bridge
CMake Error at cmake/find_ros1_package.cmake:29 (message):
  find_ros1_package() failed to find 'tf' using pkg_check_modules()
Call Stack (most recent call first):
  CMakeLists.txt:60 (find_ros1_package)

---
Failed   <<< ros1_bridge [5.15s, exited with code 1]

Summary: 0 packages finished [6.75s]
  1 package failed: ros1_bridge
  1 package had stderr output: ros1_bridge

real    0m7.928s
user    0m5.017s
sys     0m2.876s
```

Ambrose

TommyChangUMD, 2024-01-22:
Hi ambrosekwok,

I have updated Dockerfile to add support for AMD64 cpus.  To enable it, replace the first line with:

`FROM arm64v8/ros:humble-ros-base-jammy`

ambrosekwok, 2024-01-22:
Thank you,

I tried the updated Dockerfile with `FROM arm64v8/ros:humble-ros-base-jammy`, but I still got the same errors while building  the ros1_bridge package.

```
--   Package 'message_generation', required by 'tf2_msgs', not found
CMake Error at cmake/find_ros1_package.cmake:29 (message):
  find_ros1_package() failed to find 'tf' using pkg_check_modules()
Call Stack (most recent call first):
  CMakeLists.txt:60 (find_ros1_package)

-- Configuring incomplete, errors occurred!
See also "/ros-humble-ros1-bridge/build/ros1_bridge/CMakeFiles/CMakeOutput.log".
--- stderr: ros1_bridge
CMake Error at cmake/find_ros1_package.cmake:29 (message):
  find_ros1_package() failed to find 'tf' using pkg_check_modules()
Call Stack (most recent call first):
  CMakeLists.txt:60 (find_ros1_package)

---
Failed   <<< ros1_bridge [4.35s, exited with code 1]
```

ambrosekwok, 2024-01-22:
Yes, I use the latest Dockerfile provided and only modified the first line for arm64 base image.

And I tried in jetson orin nx and m2 mac mini, both failed to build the image.

TommyChangUMD, 2024-01-22:
OK.  In that case, can you post the complete docker build output?

ambrosekwok, 2024-01-23:
Thanks,

Here is the docker build output.

[build.log](https://github.com/TommyChangUMD/ros-humble-ros1-bridge-builder/files/14017453/build.log)

TommyChangUMD, 2024-01-23:
Thanks a lot!    I have updated Dockerfile.    It should work this time.  You can use the Dockerfile as-is.  There is no need to replace the first line as the base image is multi-platform already.

ambrosekwok, 2024-01-23:
Unfortunately, this still not work on my jetson and m2 mac mini.

[build.log](https://github.com/TommyChangUMD/ros-humble-ros1-bridge-builder/files/14019021/build.log)

TommyChangUMD, 2024-01-23:
OK.  I was hoping that using the binary packages would solve the issue.  At this point, we would need to dig deeper.   Can you try the following:

0.) Temporary comment out steps 8 and 9 in the Dockerfile so that the docker image builds.

1.) Log into the created docker container
`$ docker run -it --rm ros-humble-ros1-bridge-builder bash
`

While inside the container, do the following:
2.) Check to see if tf2_msgs-related packages are installed
```
$ dpkg -l | grep libtf2
libtf2-2d:amd64
libtf2-bullet-dev:amd64
libtf2-dev:amd64
libtf2-eigen-dev:amd64
libtf2-geometry-msgs-dev:amd64
libtf2-kdl-dev:amd64
libtf2-msgs-dev:amd64
libtf2-ros-dev:amd64
libtf2-ros1d:amd64
libtf2-sensor-msgs-dev:amd64
```

3.) Check to see if the the ft2-msgs cmake files are installed
```
$ ls /usr/share/tf2_msgs/
action  cmake  msg  package.xml  srv

$ ls /usr/share/tf2_msgs/cmake/tf2_msgsConfig.cmake 
/usr/share/tf2_msgs/cmake/tf2_msgsConfig.cmake
```

4.) Check to see if the the ft2-msgs configure setting can be found
```
$ pkg-config --list-all | grep tf2
tf2                       tf2 - Description of tf2
tf2_bullet                tf2_bullet - Description of tf2_bullet
tf2_eigen                 tf2_eigen - Description of tf2_eigen
tf2_geometry_msgs         tf2_geometry_msgs - Description of tf2_geometry_msgs
tf2_kdl                   tf2_kdl - Description of tf2_kdl
tf2_msgs                  tf2_msgs - Description of tf2_msgs
tf2_py                    tf2_py - Description of tf2_py
tf2_ros                   tf2_ros - Description of tf2_ros
tf2_sensor_msgs           tf2_sensor_msgs - Description of tf2_sensor_msgs
tf2_tools                 tf2_tools - Description of tf2_too
```

4.) Check the ft2-msgs configure setting
```
$ pkg-config --modversion tf2_msgs
0.7.5
```

TommyChangUMD, 2024-01-23:
From the log, it says:
  `Package 'message_generation', required by 'tf2_msgs', not found`

So, it looks like `tf2_msgs` may not be the issue.  The issue is that `message_generation` is missing.
Make sure `ros-message-generation` is installed.   I've updated the Dockerfile, please try again let me know if it works.

ambrosekwok, 2024-01-24:
Checked the `ros-message-generation` was installed with `ros-core-dev` or `ros-robot-dev`. 
But  `pkg-config --list-all | grep message_generation` show nothing.

And I found that some pkg-config files seems located in wrong directory on the jetson.

```
$ ls /usr/lib/x86_64-linux-gnu/pkgconfig/
eigen_stl_containers.pc  gencpp.pc  genlisp.pc  message_generation.pc  roslisp.pc
```

I copied  the files to `/usr/lib/aarch64-linux-gnu/pkgconfig/` and `pkg-config` can find it now.
```
$ cp /usr/lib/x86_64-linux-gnu/pkgconfig/message_generation.pc /usr/lib/aarch64-linux-gnu/pkgconfig/
$ pkg-config --list-all | grep message_generation
message_generation        message_generation - Description of message_generation
```

Able to build ros1_bridge with tf message types after that.
```
$ ros2 run ros1_bridge dynamic_bridge --print-pairs | grep tf
  - 'tf2_msgs/msg/TF2Error' (ROS 2) <=> 'tf2_msgs/TF2Error' (ROS 1)
  - 'tf2_msgs/msg/TFMessage' (ROS 2) <=> 'tf2_msgs/TFMessage' (ROS 1)
  - 'tf2_msgs/msg/TFMessage' (ROS 2) <=> 'tf/tfMessage' (ROS 1)
  - 'tf2_msgs/srv/FrameGraph' (ROS 2) <=> 'tf2_msgs/FrameGraph' (ROS 1)
```

TommyChangUMD, 2024-01-24:
I've updated Dockerfile to include the fix you found.  Could you test it?   Thanks.

ambrosekwok, 2024-01-24:
The updated Dockerfile work, thank you.
Also checked I can receive topic with `ros2 topic echo /tf`.

Thanks a lot.

closed 2024-01-24 by ambrosekwok