TommyChangUMD/ros-humble-ros1-bridge-builder #6: How to bridge grid_map_msgs?

Codaero, 2024-01-24:
I'm trying to bridge grid_map messages from ROS1 Noetic to ROS2 Humble.

The library is here:
https://github.com/anybotics/grid_map

Any ideas how I can do this? I am keep getting this error:

```
failed to create 1to2 bridge for topic '/elevation_mapping/elevation_map_raw' with ROS 1 type 'grid_map_msgs/GridMap' and ROS 2 type 'grid_map_msgs/msg/GridMap': No template specialization for the pair
```

TommyChangUMD, 2024-01-24:
Hi Codaero,

Looks like the `grid_map_msgs` needs to be added.  I will take a closer look later this week.

TommyChangUMD, 2024-01-24:
I have created a `grid-map` branch for testing.  So, just do:
```
git remote update 
git checkout grid-map
```
And then build the docker image as usual.  By the way, your ROS Humble needs to have `ros-humble-grid-map` installed.

Bridge seems to work with grid-map now:
```
$ ros2 run ros1_bridge dynamic_bridge  --print-pairs | grep grid_map
  - 'grid_map_msgs/msg/GridMap' (ROS 2) <=> 'grid_map_msgs/GridMap' (ROS 1)
  - 'grid_map_msgs/msg/GridMapInfo' (ROS 2) <=> 'grid_map_msgs/GridMapInfo' (ROS 1)
  - 'grid_map_msgs/srv/GetGridMap' (ROS 2) <=> 'grid_map_msgs/GetGridMap' (ROS 1)
  - 'grid_map_msgs/srv/GetGridMapInfo' (ROS 2) <=> 'grid_map_msgs/GetGridMapInfo' (ROS 1)
  - 'grid_map_msgs/srv/ProcessFile' (ROS 2) <=> 'grid_map_msgs/ProcessFile' (ROS 1)
  - 'grid_map_msgs/srv/SetGridMap' (ROS 2) <=> 'grid_map_msgs/SetGridMap' (ROS 1)

```

Codaero, 2024-01-24:
Hi Tommy, thanks so much for the help, will give it a try and will see if there are any issues.

TommyChangUMD, 2024-01-24:
No problem.  Alternatively, you can now also use the main branch.  But build the image by:

`docker build . --build-arg ADD_grid_map=1 -t ros-humble-ros1-bridge-builder`

Codaero, 2024-01-25:
Everything looks like it is working so far. Excellent documentation by the way.

closed 2024-01-25 by Codaero