TommyChangUMD/ros-humble-ros1-bridge-builder #8: Cannot find shared libraries

Codaero, 2024-02-07:
I am trying to run the ros1-bridge on a new computer and I am unfortunately getting a "cannot open shared object file" error:

```code
codaero@codaero-Katana-17-B13VGK:~/ros-humble-ros1-bridge$ ros2 run ros1_bridge dynamic_bridge
/home/codaero/ros-humble-ros1-bridge/install/ros1_bridge/lib/ros1_bridge/dynamic_bridge: error while loading shared libraries: libroscpp.so.4d: cannot open shared object file: No such file or directory
[ros2run]: Process exited with failure 127
```

What may be causing this?

TommyChangUMD, 2024-02-08:
hmm..  Did you source the overlay first?
`source ros-humble-ros1-bridge/install/local_setup.bash`

The libroscpp.so.4d file should be unpacked already and it's located at:

`ros-humble-ros1-bridge/install/ros1_bridge/lib/libroscpp.so.4d`

Codaero, 2024-02-08:
Yes, I did the following when setting it up on a new PC:

```code
source /opt/ros/humble/setup.bash
source ros-humble-ros1-bridge/install/local_setup.bash
```

TommyChangUMD, 2024-02-08:
Thank you for report this bug.  I've fixed it now.  
So, please pull the latest Dockerfile and try again.

Codaero, 2024-02-11:
this works now thanks

closed 2024-02-11 by Codaero