TommyChangUMD/ros-humble-ros1-bridge-builder #9: ROS1_bridge Fails to run

Bexprove, 2024-02-20:
rakshith@rakshith-ROG:~$ ros2 run ros1_bridge dynamic_bridge
/home/rakshith/ros-humble-ros1-bridge/install/ros1_bridge/lib/ros1_bridge/dynamic_bridge: error while loading shared libraries: libgrid_map_msgs__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory
[ros2run]: Process exited with failure 127

I get the following error while running. I am using ubuntu 22.04 with ros2 humble.

TommyChangUMD, 2024-02-20:
Most likely you don't have grid_map installed on your ROS 2 Humble system.  Try:

`sudo apt -y install ros-humble-grid-map`
or
`sudo apt -y install ros-humble-grid-map-msgs`

TommyChangUMD, 2024-02-20:
By the way, if you don't need grid-map support, then don't use the `--build-arg ADD_grid_map=1` option when building the docker image.

TommyChangUMD, 2024-02-20:
I have modified README.md to clarify the build options.

Bexprove, 2024-02-20:
thank you, this did fix my issue, I think this can be closed then

closed 2024-02-20 by Bexprove