TommyChangUMD/ros-humble-ros1-bridge-builder #10: Missing example_interfaces shared object

Codaero, 2024-02-29:
Recently tried to build the docker image and run the ros1 bridge on a new robot w/ROS2 Humble. I encountered the following error:

```
/home/administrator/ros-humble-ros1-bridge/install/ros1_bridge/lib/ros1_bridge/dynamic_bridge: error while loading shared libraries: libexample_interfaces__rosidl_typesupport_cpp.so: cannot open shared object file: No such file or directory
[ros2run]: Process exited with failure 127
```

Is this something to do with the example interfaces?

TommyChangUMD, 2024-03-01:
Hi Codaero,

Look like you don't have the `ros-humble-example-interfaces` package installed.
It's included in the `ros-humble-desktop` package.  So, I suggest you install in on your ROS2 Humble system:

      `apt-get install ros-humble-desktop`

Codaero, 2024-03-01:
My apologies, thanks for clarifying I appreciate it. I assumed the system I was working on already had humble-desktop but apparently it did not.

closed 2024-03-01 by Codaero