TommyChangUMD/ros-humble-ros1-bridge-builder #11: Custom Message Type Support

Codaero, 2024-03-21:
Any idea on how to incorporate bridging custom message types?

TommyChangUMD, 2024-03-22:
Hi Robert,

Bridging a custom message type is straightforward. Firstly, we'll need the source code of the custom message for both ROS1 and ROS2. Then, we add an instruction to compile the ROS1 version of the custom message to the bridge builder Dockerfile. That's all for the ROS1 part (See steps 6.1, 6.2, and 7 in the Dockerfile for examples).

Now, on the ROS2 system, we simply compile the custom message type (and source its setup.bash overlay afterward) as usual . When we run the ROS1 bridge, the custom message type should automatically get bridged.  To verify that, run the `ros2 run ros1_bridge dynamic_bridge --print-pairs` command and look for the custom message in the list.

That's the gist of it. Do you have the source code of the custom message type you wish to bridge? If you can share it, I can put everything together and post it.

aserbremen, 2024-03-25:
Hey, some custom messages for me are not properly configured.

Inside the bridge builder docker file I add some local custom msgs at step 6.1 like so: 

```
# add additional ROS1 custom messages
RUN mkdir -p /ros1_ws_msgs/src
COPY catkin_ws/src/others/robotnik_msgs /ros1_ws_msgs/src/robotnik_msgs
RUN cd /ros1_ws_msgs && unset ROS_DISTRO && \
     time colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```
When entering the ros-humble-ros-bridge docker container I can `source /ros1_ws_msgs/install/setup.bash` and see with `rosmsg list` that the custom messages are available. 

At step 7 I add the additional sourcing of the ros1 msg folder `source /ros1_ws_msgs/install/setup.bash &&...` before sourcing ros humble. 

However, when sourcing the custom ROS2 messages on the ROS2 host system, `ros2 run ros1_bridge dynamic_bridge --print-pairs` doesn't register the pairs. I assume during the ros builder bridge build stage, the custom messages need to be available in ROS1 and ROS2, but I am not sure.

aserbremen, 2024-03-25:
After adding the custom messages also as a ros2 package during the build stage of the bridge builder Dockerfile

```
# 6.1 Also add the ROS2 custom messages
RUN mkdir -p /ros2_ws_msgs/src
COPY ros2_ws_msgs/src/robotnik_msgs /ros2_ws_msgs/src/robotnik_msgs
RUN cd /ros2_ws_msgs && \
     source /opt/ros/humble/setup.bash && \
     time colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
```
and sourcing both the ROS1 and the ROS2 custom message package at step 7 it worked 
```
# Step 7. Compile ros1_bridge
...
source /ros1_ws_msgs/install/setup.bash && \
source /opt/ros/humble/setup.bash  && \
source /ros2_ws_msgs/install/local_setup.bash &&  \ 
...
```

Thanks for this repo! Very helpful

closed 2024-03-25 by TommyChangUMD

Codaero, 2024-03-29:
I am still having issues with not seeing the custom message pair when I do 

`source /opt/ros/humble/setup.bash` followed by
`source ~/ros-humble-ros1-bridge/install/local_setup.bash` followed by 
`ros2 run ros1_bridge dynamic_bridge --print-pairs`

I am still not getting the custom message pair to show up. When I built the Docker image and went inside the Docker container, sourcing my `ros1_ws_msgs/install/setup.bash` and `ros2_ws_msgs/install/setup.bash` I was able to view the custom message using `rosmsg list` and `ros2 interface list` just fine.

This is a GitHub repo containing the ros1 and ros2 messages: https://github.com/Codaero/custom_msgs

This is my Dockerfile for reference:

```
FROM ros:humble-ros-base-jammy

SHELL ["/bin/bash", "-o", "pipefail", "-o", "errexit", "-c"]

###########################
# 1.) Bring system up to the latest ROS desktop configuration
###########################
RUN apt-get -y update
RUN apt-get -y upgrade
RUN apt-get -y install ros-humble-desktop

###########################
# 2.) Temporarily remove ROS2 apt repository
###########################
RUN mv /etc/apt/sources.list.d/ros2-latest.list /root/
RUN apt-get update

###########################
# 3.) comment out the catkin conflict
###########################
RUN sed  -i -e 's|^Conflicts: catkin|#Conflicts: catkin|' /var/lib/dpkg/status
RUN apt-get install -f

###########################
# 4.) force install these packages
###########################
RUN apt-get download python3-catkin-pkg
RUN apt-get download python3-rospkg
RUN apt-get download python3-rosdistro
RUN dpkg --force-overwrite -i python3-catkin-pkg*.deb
RUN dpkg --force-overwrite -i python3-rospkg*.deb
RUN dpkg --force-overwrite -i python3-rosdistro*.deb
RUN apt-get install -f

###########################
# 5.) Install the latest ROS1 desktop configuration
# see https://packages.ubuntu.com/jammy/ros-desktop-dev
# note: ros-desktop-dev automatically includes tf tf2
###########################
RUN apt-get -y install ros-desktop-dev

# fix ARM64 pkgconfig path issue -- Fix provided by ambrosekwok 
RUN if [[ $(uname -m) = "arm64" || $(uname -m) = "aarch64" ]]; then                    \
    cp /usr/lib/x86_64-linux-gnu/pkgconfig/* /usr/lib/aarch64-linux-gnu/pkgconfig/;    \
    fi

###########################
# 6.) Restore the ROS2 apt repos
###########################
RUN mv /root/ros2-latest.list /etc/apt/sources.list.d/
RUN apt-get -y update

# Add custom messages ROS1
RUN mkdir -p /ros1_ws_msgs/src
COPY custom_msgs_ros1 /ros1_ws_msgs/src/custom_msgs_ros1
RUN cd /ros1_ws_msgs && unset ROS_DISTRO && \
     time colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
[... 48 lines cut ...]
     ROS1_LIBS="$ROS1_LIBS libroscpp.so";                                               \
     ROS1_LIBS="$ROS1_LIBS libroscpp_serialization.so";                                 \
     ROS1_LIBS="$ROS1_LIBS librosconsole.so";                                           \
     ROS1_LIBS="$ROS1_LIBS librosconsole_log4cxx.so";                                   \
     ROS1_LIBS="$ROS1_LIBS librosconsole_backend_interface.so";                         \
     ROS1_LIBS="$ROS1_LIBS liblog4cxx.so";                                              \
     ROS1_LIBS="$ROS1_LIBS libcpp_common.so";                                           \
     ROS1_LIBS="$ROS1_LIBS libb64.so";                                                  \
     ROS1_LIBS="$ROS1_LIBS libaprutil-1.so";                                            \
     ROS1_LIBS="$ROS1_LIBS libapr-1.so";                                                \
     ROS1_LIBS="$ROS1_LIBS libactionlib.so.1d";                                         \
     cd /ros-humble-ros1-bridge/install/ros1_bridge/lib;                                \
     for soFile in $ROS1_LIBS; do                                                       \
         soFilePath=$(ldd libros1_bridge.so | grep $soFile | awk '{print $3;}');        \
         cp $soFilePath ./;                                                             \
     done

###########################
# 10.) Spit out ros1_bridge tarball by default when no command is given
###########################
RUN tar czf /ros-humble-ros1-bridge.tgz                                 \
     --exclude '*/build/*' --exclude '*/src/*' /ros-humble-ros1-bridge 
ENTRYPOINT []
CMD cat /ros-humble-ros1-bridge.tgz; sync
```