TommyChangUMD/ros-humble-ros1-bridge-builder #12: Custom Message Bridging Issues Still

Codaero, 2024-03-29:
I am still having issues with not seeing the custom message pair when I do 

`source /opt/ros/humble/setup.bash` followed by
`source ~/ros-humble-ros1-bridge/install/local_setup.bash` followed by 
`ros2 run ros1_bridge dynamic_bridge --print-pairs`

I am still not getting the custom message pair to show up. When I built the Docker image and went inside the Docker container, sourcing my `ros1_ws_msgs/install/setup.bash` and `ros2_ws_msgs/install/setup.bash` I was able to view the custom message using `rosmsg list` and `ros2 interface list` just fine.

This is a GitHub repo containing the ros1 and ros2 messages: https://github.com/Codaero/custom_msgs

This is my Dockerfile for reference:

```
FROM ros:humble-ros-base-jammy

SHELL ["/bin/bash", "-o", "pipefail", "-o", "errexit", "-c"]

###########################
# 1.) Bring system up to the latest ROS desktop configuration
###########################
RUN apt-get -y update
RUN apt-get -y upgrade
RUN apt-get -y install ros-humble-desktop

###########################
# 2.) Temporarily remove ROS2 apt repository
###########################
RUN mv /etc/apt/sources.list.d/ros2-latest.list /root/
RUN apt-get update

###########################
# 3.) comment out the catkin conflict
###########################
RUN sed  -i -e 's|^Conflicts: catkin|#Conflicts: catkin|' /var/lib/dpkg/status
RUN apt-get install -f

###########################
# 4.) force install these packages
###########################
RUN apt-get download python3-catkin-pkg
RUN apt-get download python3-rospkg
RUN apt-get download python3-rosdistro
RUN dpkg --force-overwrite -i python3-catkin-pkg*.deb
RUN dpkg --force-overwrite -i python3-rospkg*.deb
RUN dpkg --force-overwrite -i python3-rosdistro*.deb
RUN apt-get install -f

###########################
# 5.) Install the latest ROS1 desktop configuration
# see https://packages.ubuntu.com/jammy/ros-desktop-dev
# note: ros-desktop-dev automatically includes tf tf2
###########################
RUN apt-get -y install ros-desktop-dev

# fix ARM64 pkgconfig path issue -- Fix provided by ambrosekwok 
RUN if [[ $(uname -m) = "arm64" || $(uname -m) = "aarch64" ]]; then                    \
    cp /usr/lib/x86_64-linux-gnu/pkgconfig/* /usr/lib/aarch64-linux-gnu/pkgconfig/;    \
    fi

###########################
# 6.) Restore the ROS2 apt repos
###########################
RUN mv /root/ros2-latest.list /etc/apt/sources.list.d/
RUN apt-get -y update

# Add custom messages ROS1
RUN mkdir -p /ros1_ws_msgs/src
COPY custom_msgs_ros1 /ros1_ws_msgs/src/custom_msgs_ros1
RUN cd /ros1_ws_msgs && unset ROS_DISTRO && \
     time colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release
     
# Add custom messages ROS2
RUN mkdir -p /ros2_ws_msgs/src
COPY custom_msgs_ros2 /ros2_ws_msgs/src/custom_msgs_ros2
RUN cd /ros2_ws_msgs && \
     source /opt/ros/humble/setup.bash && \
     time colcon build --cmake-args -DCMAKE_BUILD_TYPE=Release

# unit test (optional)
RUN if [[ "$ADD_ros_tutorials" = "1" ]]; then           \
     cd ros_tutorials && unset ROS_DISTRO &&            \
     colcon test --event-handlers console_direct+ &&    \
     colcon test-result;                                \
     fi

###########################
# 7.) Compile ros1_bridge
###########################
RUN  source /ros1_ws_msgs/install/setup.bash && 					\
     source /opt/ros/humble/setup.bash  &&                                              \
     source /ros2_ws_msgs/install/local_setup.bash &&  					\
     mkdir -p /ros-humble-ros1-bridge/src &&                                            \
     cd /ros-humble-ros1-bridge/src &&                                                  \
     git clone https://github.com/smith-doug/ros1_bridge.git &&                         \
     cd ros1_bridge/ &&                                                                 \
     git checkout action_bridge_humble &&                                               \
     cd ../.. &&                                                                        \
     MEMG=$(printf "%.0f" $(free -g | awk '/^Mem:/{print $2}'));                        \
     NPROC=$(nproc);  MIN=$((MEMG<NPROC ? MEMG : NPROC));                               \
     echo "Please wait...  running $MIN concurrent jobs to build ros1_bridge" &&        \
     time MAKEFLAGS="-j $MIN" colcon build --event-handlers console_direct+             \
       --cmake-args -DCMAKE_BUILD_TYPE=Release 

###########################
# 8.) Clean up
###########################
RUN apt-get -y clean all; apt-get -y update

###########################
# 9.) Pack all ROS1 dependent libraries
###########################
# fix ARM64 pkgconfig path issue -- Fix provided by ambrosekwok 
RUN if [[ $(uname -m) = "arm64" || $(uname -m) = "aarch64" ]]; then                    \
    cp /usr/lib/x86_64-linux-gnu/pkgconfig/* /usr/lib/aarch64-linux-gnu/pkgconfig/;    \
    fi

RUN ROS1_LIBS="libxmlrpcpp.so";                                                         \
     ROS1_LIBS="$ROS1_LIBS librostime.so";                                              \
     ROS1_LIBS="$ROS1_LIBS libroscpp.so";                                               \
     ROS1_LIBS="$ROS1_LIBS libroscpp_serialization.so";                                 \
     ROS1_LIBS="$ROS1_LIBS librosconsole.so";                                           \
     ROS1_LIBS="$ROS1_LIBS librosconsole_log4cxx.so";                                   \
     ROS1_LIBS="$ROS1_LIBS librosconsole_backend_interface.so";                         \
     ROS1_LIBS="$ROS1_LIBS liblog4cxx.so";                                              \
     ROS1_LIBS="$ROS1_LIBS libcpp_common.so";                                           \
     ROS1_LIBS="$ROS1_LIBS libb64.so";                                                  \
     ROS1_LIBS="$ROS1_LIBS libaprutil-1.so";                                            \
     ROS1_LIBS="$ROS1_LIBS libapr-1.so";                                                \
     ROS1_LIBS="$ROS1_LIBS libactionlib.so.1d";                                         \
     cd /ros-humble-ros1-bridge/install/ros1_bridge/lib;                                \
     for soFile in $ROS1_LIBS; do                                                       \
         soFilePath=$(ldd libros1_bridge.so | grep $soFile | awk '{print $3;}');        \
         cp $soFilePath ./;                                                             \
     done

###########################
# 10.) Spit out ros1_bridge tarball by default when no command is given
###########################
RUN tar czf /ros-humble-ros1-bridge.tgz                                 \
     --exclude '*/build/*' --exclude '*/src/*' /ros-humble-ros1-bridge 
ENTRYPOINT []
CMD cat /ros-humble-ros1-bridge.tgz; sync
```

_Originally posted by @Codaero in https://github.com/TommyChangUMD/ros-humble-ros1-bridge-builder/issues/11#issuecomment-2027676441_

TommyChangUMD, 2024-03-29:
Hi Robert,

Thanks for reporting.  Let me poke around to see what might be the issue.

TommyChangUMD, 2024-03-30:
Robert, it turns out you just need to make sure the package name ends with "_msgs".   And if you use the same package names for both ROS1 and ROS2 packages, everything will just work.  

I have cloned your repo and made those changes.  https://github.com/TommyChangUMD/custom_msgs/tree/main
And I went ahead and put everything together in the latest Dockerfile and updated the README.md file accordingly.  

You will need to use the `--build-arg ADD_custom_msgs=1` option to try it.

TommyChangUMD, 2024-03-30:
The link below explains how ROS 1 and 2 messages are associated with each other.

https://github.com/ros2/ros1_bridge/blob/master/doc/index.rst

TommyChangUMD, 2024-04-06:
Hi Robert,

I am guessing that you were able to get everything working now.  If not, feel free to open another issue.

closed 2024-04-06 by TommyChangUMD