TommyChangUMD/ros-humble-ros1-bridge-builder #13: process has died [version `GLIBC_2.33' not found]

Chandan-kumar-R, 2024-04-12:
Hello, 
Thank you so much for this repo. While running i get the follow error:
```
/opt/ros/noetic/lib/rosout/rosout: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /home/dragon/ros-humble-ros1-bridge/install/ros1_bridge/lib/libapr-1.so.0)
[rosout-1] process has died [pid 2098, exit code 1, cmd /opt/ros/noetic/lib/rosout/rosout __name:=rosout __log:=/home/dragon/.ros/log/da3a8dd0-f89e-11ee-81b3-d4e98a419308/rosout-1.log].
log file: /home/dragon/.ros/log/da3a8dd0-f89e-11ee-81b3-d4e98a419308/rosout-1*.log
```
i have tried updating my system:` sudo apt-get update && sudo apt-get upgrade`
and also tried to install glibc:  `sudo apt install glibc`  but error remain same.

System: Ubuntu 22.04

Any assistance in resolving this issue would be greatly appreciated. Thank you.

TommyChangUMD, 2024-04-12:
hmm.  I see you are running `roscore` in a ROS Noetic system.   I assume this ROS Noetic system is running on a Ubuntu 20.04 or older?   Anyways, it shouldn't interfere with what you have installed on your Ubuntu 22.04.    Right?   

If you have both ROS humble and ROS Noetic installed on Ubuntu 22.04.  You are going to get conflict like this.   You are better off running ROS Noetic on a Docker container or another separated machine.

Chandan-kumar-R, 2024-04-12:
> hmm. I see you are running `roscore` in a ROS Noetic system. I assume this ROS Noetic system is running on a Ubuntu 20.04 or older? Anyways, it shouldn't interfere with what you have installed on [...]

I don't have 20.4 in my system. I'm running roscore in gui terminal (using rocker command in readme file)

TommyChangUMD, 2024-04-12:
OK. I see the issue..  rocker is starting a container but sharing your Ubuntu 22.04's home directory (due to the `--home` option).  As a result, your ~/.bashrc is executed inside the container.  And it most likely you sourced the bridge's `local_setup.bash` in your ~/.bashrc.   The solution is to remove the `--home` option from the rocker command:

`rocker --x11 --user --privileged \
         --volume /dev/shm /dev/shm --network=host -- osrf/ros:noetic-desktop \
         'bash -c "sudo apt update; sudo apt install -y tilix; tilix"'`

I have updated the README.md file.  Thank you for reporting this issue.

Chandan-kumar-R, 2024-04-12:
> OK. I see the issue.. rocker is starting a container but sharing your Ubuntu 22.04's home directory (due to the `--home` option). As a result, your ~/.bashrc is executed inside the container. And it [...]

Thank you. it works now

closed 2024-04-12 by Chandan-kumar-R