TommyChangUMD/ros-humble-ros1-bridge-builder #14: performance

rdinoff, 2024-05-15:
Everything built properly but it runs very slow.   Is this your experience also?

TommyChangUMD, 2024-05-15:
You mean the building process is slow?  
The build performance depends on the amount of memory and the number of CPU cores.  

To compile the ros1 bridge, It takes about 1 GB of memory per CPU core.    So, if your system has only 4GB of memory, but 100 CPU cores, it will still use only 4 CPU cores for the compilation.

rdinoff, 2024-05-15:
sorry I meant actually running the bridge (compile time not an issue).   I know not your code, but was just curious if you see very slow run time performance?  I have an Image message topic from ros1 at 15 hz, and barely gets to ros2 at 2hz

TommyChangUMD, 2024-05-16:
I see.  You can try using the original ros1 bridge code.  Modify `Dockerfile` line 222:
Replace:
    `git clone https://github.com/smith-doug/ros1_bridge.git;`
with
    `git clone https://github.com/ros2/ros1_bridge.git;`

And the rebuild the docker image.

TommyChangUMD, 2024-08-29:
Closing this issue due to inactivity.

closed 2024-08-29 by TommyChangUMD