TommyChangUMD/ros-humble-ros1-bridge-builder #15: Custom action issue

allstarry, 2024-08-29:
Hello,
I'm trying to use ros1_bridge to communicate with a noetic action client and a ros2 action server, so I followed your instruction to build this package with my own interface package. But an error occured when I used noetic action client to send a goal to ros2 action server in a docker container.
Here's the output:
`docker-robot@b4fd266d8324:~/ros2_iron_ws$ ros2 run ros1_bridge dynamic_bridge
created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
[INFO] [1724920351.520968669] [ros_bridge]: Passing message from ROS 2 rcl_interfaces/msg/Log to ROS 1 rosgraph_msgs/Log (showing msg only once per type)
Created 1 to 2 bridge for service /lifecycle_manager_navigation/is_active
Created 1 to 2 bridge for action /take_elevator
[INFO] [1724920356.928231830] [ros_bridge]: Sending goal /axclient_912603_1724905438177-8-1724920356.927
terminate called after throwing an instance of 'rclcpp_action::exceptions::UnknownGoalHandleError'
  what():  Goal handle is not known to this client.
[ros2run]: Aborted
`

TommyChangUMD, 2024-08-29:
Hi, 

I think more information is needed.  Can you share your code so that I can try it  myself?

allstarry, 2024-09-18:
Hi, sorry for the delay.
I have confirmed that the error occurred due to my ros2 action server didn't launch successfully, and I think those log message mislead me.

TommyChangUMD, 2024-09-19:
No problem.  I am glad you got the action server working.

closed 2024-09-19 by TommyChangUMD