TommyChangUMD/ros-humble-ros1-bridge-builder #16: sensor_msgs/msgImage topic not bridged

jdcast, 2024-09-18:
Should this package bridge image topics of type `sensor_msgs/msg/Image`?

I'm not seeing any bridged when I try: `ros2 run ros1_bridge dynamic_bridge`

TommyChangUMD, 2024-09-19:
It should, if you see the pair listed when you run `ros2 run ros1_bridge dynamic_bridge --print-pairs` .   
Let me try it myself later this week and get back to you.

jdcast, 2024-09-19:
Okay thanks @TommyChangUMD . 
For context, the bridge I'm trying to form is between the following:

- Jetson Orin with ROS2 Humble running zed-ros2-wrapper (https://github.com/stereolabs/zed-ros2-wrapper) and ros-humble-ros1-bridge-builder
- Raspberry PI with ROS1 Noetic
- PC with ROS1 Noetic

I'm pretty sure I have the networking configured correctly as I see a few basic topics bridged such as `/rosout`.
Ultimately, I want to see all topics shared among all machines.
But my immediate goal is to be able to see the zed wrapper's image topics on the rpi and laptop.

jdcast, 2024-09-19:
Hi @TommyChangUMD 

I was able to bridge `sensor_msgs/msg/Image (ros2)<-> sensor_msgs/Image (ros1) ` (by default) by specifying the following flag (the images are being published from my ROS2 machine):
`ros2 run ros1_bridge dynamic_bridge --bridge-all-topics` or `ros2 run ros1_bridge dynamic_bridge --bridge-all-2to1-topics`

closed 2024-09-19 by jdcast