TommyChangUMD/ros-humble-ros1-bridge-builder #18: Noetic and Humble on different machines

dyjin2, 2024-10-07:
Hi Tommy,

Thanks for sharing this work. I'm new to Docker, and although I suspect this issue might be related to network configuration or ethernet port setting, I'm not sure how to search for a solution. I hope you can provide some insight.

My question is I want to talk to my mobile platform (Clearpath Boxer) running on ubuntu20 + ros1 noetic, from my PC running on ubuntu22 + ros2 humble, simply publishing /cmd_vel for now. I've confirmed that after setting the mobile platform as the master, the Noetic inside Docker can both read rostopics and publish to /cmd_vel. I've also confirmed that Humble on the PC and Noetic inside Docker can communicate via the bridge. But when I tried to connect all three together, I encountered this error: 

~$ ros2 run ros1_bridge dynamic_bridge 
[ERROR] [1728311155.919384649]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

Any suggestions would be greatly appreciated, thanks!

P.S. Not sure how to implement "Note1: It's important to share the host's network and the /dev/shm/ directory with the container."

TommyChangUMD, 2024-10-08:
Hi David,

It sounds like a networks issue.  But I could be wrong.  
Let me try a few things myself and get back to you.

--Tommy

TommyChangUMD, 2024-10-12:
The issue is that the bridge couldn't find the ros master at `[localhost:11311]`.  So, all you need to do is to set the `ROS_MASTER_URI` environment variable to point to the IP address and the port number of your ros master. 

So, assuming your Noetic machine's IP address is 192.168.1.208.  

Instead of:

`~$ ros2 run ros1_bridge dynamic_bridge`

You would do:

`ROS_MASTER_URI='http://192.168.1.208:11311' ros2 run ros1_bridge dynamic_bridge`

I have updated the README.md file to include a section about it:
https://github.com/TommyChangUMD/ros-humble-ros1-bridge-builder?tab=readme-ov-file#how-to-make-it-work-with-ros1-master-running-on-a-different-machine

dyjin2, 2024-10-18:
Thanks for testing this out! With the ROS_MASTER_URI setting, the bridge can find the master now. The next issue is that the bridge attempted to contact all the topics, but most failed. I'm okay with this because the network setup is quite complex, and most of those topics are customized. I didn't expect the bridge to handle all of them. 

So I tried the parameter_bridge mentioned [here](https://github.com/ros2/ros1_bridge?tab=readme-ov-file#example-4-bridge-only-selected-topics-and-services) to bridge only the selected standard topic i.e. /cmd_vel. Since I have two ROS1 machines, I've tried running 
``` bash
rosparam load bridge.yaml
```
with this bridge.yaml:
``` bash
topics:
  -
    topic: /cmd_vel  # Topic name on both ROS 1 and ROS 2
    type: geometry_msgs/msg/Twist  # Type of topic to bridge
    queue_size: 5  # Queue size
```
in both of them. Running this in my robot-ros1 returned errors, but running in docker-ros1 works (don't why it should be run in this ros1 instead of where the master is), and the output looks like this:
 ``` bash
~$ ROS_MASTER_URI='http://192.168.131.1:11311' ros2 run ros1_bridge parameter_bridge
Trying to create bidirectional bridge for topic '/cmd_vel' with ROS 2 type 'geometry_msgs/msg/Twist'
[INFO] [1729189047.943439759] [ros_bridge]: create bidirectional bridge for topic /cmd_vel
The parameter 'services_1_to_2' either doesn't exist or isn't an array
The parameter 'services_2_to_1' either doesn't exist or isn't an array
```
Now my expectation is when I publish a Twist to /cmd_vel on my ros2, the robot-ros1 will simultaneously get the same message in its /cmd_vel, thus triggering the motors. But it turned out it couldn't, only the docker-ros1 got it. And when I publish to /cmd_vel on my docker-ros1, only the robot-ros1 got it whereas the ros2 didn't.

I did a parallel test on the standard topic /chatter and got the same result (ros2 --> ros1-docker | ros1-docker <--> ros1-robot (--> means publish to)).

Could you provide some insight on this?

TommyChangUMD, 2024-10-20:
That's interesting.  I will have to try it myself and get back to you.
In the meantime, without looking into the details, I wonder if the `--bridge-all-topics` option will make any difference?

So, instead of doing, 
`ROS_MASTER_URI='http://192.168.131.1:11311' ros2 run ros1_bridge`

Maybe try the below?

`ROS_MASTER_URI='http://192.168.131.1:11311' ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`

dyjin2, 2024-10-25:
Thanks! With ```--bridge-all-topics``` setting it indeed returned several successful bridges like:
```bash
created 1to2 bridge for topic '/bluetooth_teleop/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
created 1to2 bridge for topic '/bluetooth_teleop/joy' with ROS 1 type 'sensor_msgs/Joy' and ROS 2 type 'sensor_msgs/msg/Joy'
created 1to2 bridge for topic '/bms/battery_status' with ROS 1 type 'sensor_msgs/BatteryState' and ROS 2 type 'sensor_msgs/msg/BatteryState'
``` 
I wonder if that's because they use standard msg types?
But for the rest of the failed topics, it kept trying and returned errors like:
```bash
[ERROR] [1729890515.200690520]: couldn't resolve publisher host [XXXXXX]
Failed to connect to /fleet_api/fleet_bridge/set_logger_level (XXXXXX:XXXXXX)
[ERROR] [1729890515.210253378]: couldn't resolve publisher host [XXXXXX]
Failed to connect to /platform_api/platform_bridge/set_logger_level (XXXXXX:XXXXXX)
``` 
While the errors are looping, I can see those successfully bridged topics on ROS2 side. However, it didn't make a difference on the problem (ros2 --> ros1-docker | ros1-docker <--> ros1-robot) I mentioned.

TommyChangUMD, 2024-10-29:
David,

Do you have some code or instructions that I can try to duplicate these issues myself?  It's hard for me to guess what the issues might be without the code to play with.

--

dyjin2, 2024-11-05:
Thanks again for your help. It’s likely that the computer on the mobile robot had complex settings. But no worries, we've already moved on to using socket to work around this issue. Please consider this issue resolved.

closed 2024-11-05 by dyjin2