TommyChangUMD/ros-humble-ros1-bridge-builder #20: Not able to establish communication for geometry_msgs/Twist

PeisenWong, 2024-11-09:
I am running ros2 humble in my laptop(Ubuntu 22.04) and ros1 noetic in raspberry pi model 4b (Buster), I am using this repo docker to build the ros1_bridge. However, when I used my laptop as publisher to publish `geometry_msgs/msg/Twist`, which is
 `ros2 topic pub -r 10 /cmd_vel geometry_msgs/msg/Twist "linear:
  x: 1.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0"
`
and then I run `ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`
the log will be
 `...
created 2to1 bridge for topic '/cmd_vel' with ROS 2 type 'geometry_msgs/msg/Twist' and ROS 1 type ''
[INFO] [1731121524.097166280] [ros_bridge]: Passing message from ROS 2 geometry_msgs/msg/Twist to ROS 1 geometry_msgs/Twist (showing msg only once per type)
...`
It seems like it cannot determine the ROS1 type. At the rpi side, I also cannot receive the published message.

When I do in opposite way,  which I run the publisher in rpi side, the ros1_bridge terminal will show:
`...
created 1to2 bridge for topic '/cmd_vel' with ROS 1 type 'geometry_msgs/Twist' and ROS 2 type 'geometry_msgs/msg/Twist'
[INFO] [1731123313.842520272] [ros_bridge]: Passing message from ROS 1 geometry_msgs/Twist to ROS 2 geometry_msgs/msg/Twist (showing msg only once per type)
...`
In this way, my ubuntu 22.04 system can receive the published message from rpi.

In summary, my problem is that the rpi noetic cannot receive message published from ubuntu humble, but ubuntu humble can receive message from rpi noetic. Please give me some advices.

The command `ros2 run ros1_bridge dynamic_bridge --print-pairs` shows
`...
'geometry_msgs/msg/Twist' (ROS 2) <=> 'geometry_msgs/Twist' (ROS 1)
...`

Command to build docker: ` docker build . --build-arg ADD_custom_msgs=1 -t ros-humble-ros1-bridge-builder`
I followed the README and the compiling process has no problem.

TommyChangUMD, 2024-11-09:
Have you verified this is also the behavior with ros noetic running inside the docker container (instead of on a remote raspberry pi machine)?   It's working fine for me.

When I run:
      `ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`

      <I got>
        created 2to1 bridge for topic '/cmd_vel' with ROS 2 type 'geometry_msgs/msg/Twist' and ROS 1 type 'geometry_msgs/Twist'
        created 2to1 bridge for topic '/rosout' with ROS 2 type 'rcl_interfaces/msg/Log' and ROS 1 type 'rosgraph_msgs/Log'
        [INFO] [1731163481.818010430] [ros_bridge]: Passing message from ROS 2 geometry_msgs/msg/Twist to ROS 1 geometry_msgs/Twist (showing msg only once per type)
        removed 2to1 bridge for topic '/cmd_vel'
        created 2to1 bridge for topic '/cmd_vel' with ROS 2 type 'geometry_msgs/msg/Twist' and ROS 1 type 'geometry_msgs/Twist'

Also, instead of `--bridge-all-topics`, you might want to try `--bridge-all-1to2-topics` to see if that makes any difference?
I have updated the README.md to provide more information about these parameters.

PeisenWong, 2024-11-10:
Hi @TommyChangUMD .
I havent try to run in docker, will try it later. Let me update my current situation.

Same as my issue, I want to communicate from rpi noetic to ubuntu 22.04 humble.

This is my testing procedure:
I ssh into rpi and run `roscore`
Then I source custom_msgs/install/setup.bash & source ros1_bridge_builder/local_setup.bash
I am using /cmd_ve topic to test communication (No typo). I open another terminal in humble to run 
`ros2 topic pub -r 10 /cmd_ve geometry_msgs/msg/Vector3 "{x: 1.0, y: 0.0, z: 0.5}"`
Then i run `ros2 run ros1_bridge dynamic_bridge --bridge-all-topics`
Here is its output:
`...
created 2to1 bridge for topic '/cmd_ve' with ROS 2 type 'geometry_msgs/msg/Vector3' and ROS 1 type 'geometry_msgs/Vector3'
[INFO] [1731240547.070943429] [ros_bridge]: Passing message from ROS 2 geometry_msgs/msg/Vector3 to ROS 1 geometry_msgs/Vector3 (showing msg only once per type)
...` 

In rpi terminal, I run `rostopic hz /cmd_ve`
Then I get
`
subscribed to [/cmd_ve]
no new messages
no new messages
no new messages
^Cno new messages
`

I think this is quite weird, because I am able to receive the message publish from rpi noetic. My target is to publish message from humble to noetic to let it run. I need to able to receive the message first, but seems cannot. I am not sure where is the problem exist. Btw, the `--bridge-all-1to2-topics` did not work.

TommyChangUMD, 2024-11-13:
OK.  Thanks for the additional details.  Let me take a closer look and get back to you.

PeisenWong, 2024-11-15:
Hi, I just solve this issue, this is because of hostname issue.

In `/etc/hosts` of rpi, I need to specify the ip of laptop, so the rpi can receive the message from laptop, it works now.

closed 2024-11-15 by PeisenWong