TommyChangUMD/ros-humble-ros1-bridge-builder #21: Support Focal instead of Jammy for ROS 2 Humble + ROS 1 Noetic

Ryanf55, 2024-11-15:
Hello,

I'm reading REP-2000 and noticed that ROS2 humble has tier 3 support for Ubuntu focal. 
Meanwhile, ROS 1 noetic does not have any claimed support for Ubuntu Jammy.
https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025

Would you be open to a contribution for a Focal base image that accomplishes the same goal?

TommyChangUMD, 2024-11-15:
I am not sure I understand.
You want to run ros1 bridge on Ubuntu Focal (20.04) with ROS2 Humble installed?

TommyChangUMD, 2024-11-15:
I think you should be able to compile the ROS 1 bridge directly on the Ubuntu 20.04 + ROS 2 Humble setup. Have you tried compiling the ROS 1 bridge using the instructions provided at:

https://github.com/smith-doug/ros1_bridge

TommyChangUMD, 2024-11-15:
Actually, Jay has already done that.  (Installing ros1 bridge on Ubuntu 20.04 + ROS2 Humble)

https://github.com/jayprajapati009/ros1_bridge_tutorial

Ryanf55, 2024-11-16:
> How are you installing ROS2 humble on Ubuntu 20.04? From the source code? Or using robostack https://robostack.github.io/index.html ?

Source  for humble, binaries for noetic. See here for the full implementation: 
https://github.com/osrf/docker_images/pull/778

It's following the same methodology as Jay - his README is very nice. 

I was more curious why you chose to develop on Jammy instead of focal.

TommyChangUMD, 2024-11-16:
I see.  Nice work! 

And the main reason that I am using Ubuntu 22.04 (Jammy) is because it allows me to install the ROS 2 Humble binary package directly, without needing to compile it from source.

Ryanf55, 2024-11-16:
Is that to save build time? I avoided ROS 1 on Ubuntu 22 because of all the dpkg issues and shared library link problems.

Through the use of docker caching, doing a source build of humble doesn't take a lot of time. If you are interested, I could split that into a multi-stage build and push it to dockerhub.

TommyChangUMD, 2024-11-17:
Yes, but not only to save build time but also to take advantage of numerous additional pre-built binary packages, such as Gazebo ROS plugins, TurtleBot3, and others that are not an integral part of the ROS Humble source code.

Ryanf55, 2024-11-17:
Thanks, I appreciate the motivation, and think that we were after different goals.

To install the rest of these ROS 2 packages, I just use a Jammy docker environment. The only thing that really needs compiled is the dependencies of the ros bridge and its messages.

closed 2024-11-17 by Ryanf55