TommyChangUMD/ros-humble-ros1-bridge-builder #22: Bridge Humble to Indigo

toffanetto, 2024-12-11:
Greetings @TommyChangUMD! The project I am working on depends on using an old ROS1 version, the Indigo, based on Ubuntu 14.04. 

I am trying to use your repository. The bridge can see the topics and nodes in ROS1 and ROS2 on both sides, but the data through the topics doesn't flow in any direction.

Do you have any idea of how I can debug this problem? 

This is the output:

Launching ros1-bridge in ROS2 with `--bridge-all-topics` flag, and talker in ROS1, I get in ROS2 terminal:

```
  ROS 1: /chatter (std_msgs/String) [>0 pubs, 0 subs]
  ROS 1: /rosout (rosgraph_msgs/Log) [>0 pubs, >0 subs]
  ROS 1: /rosout_agg (rosgraph_msgs/Log) [>0 pubs, 0 subs]

  ROS 2: /chatter (std_msgs/msg/String) [0 pubs, 0 subs]
  ROS 2: /parameter_events (rcl_interfaces/msg/ParameterEvent) [1 pubs, 1 subs]
  ROS 2: /rosout (rcl_interfaces/msg/Log) [1 pubs, 0 subs]

[ERROR] [1733921060.917218861]: couldn't resolve publisher host [lma-ubuntu1404]
Failed to connect to /talker_2696_1733921049705/set_logger_level (lma-ubuntu1404:45666)
[ERROR] [1733921060.923248826]: couldn't resolve publisher host [lma-ubuntu1404]
Failed to connect to /talker_2696_1733921049705/get_loggers (lma-ubuntu1404:45666)
[ERROR] [1733921060.926734381]: couldn't resolve publisher host [lma-ubuntu1404]
Failed to connect to /rosout/get_loggers (lma-ubuntu1404:34969)
[ERROR] [1733921060.930316600]: couldn't resolve publisher host [lma-ubuntu1404]
``` 

Weirdly, the bridge shows >0 pubs for /chatter, and I am getting some errors with default node services, such as `/get_loggers`.

TommyChangUMD, 2024-12-12:
Just a hunch:
   `couldn't resolve publisher host [lma-ubuntu1404]`
looks like a network setting issue?  

Can you ping lma-ubuntu1404 ?   Is it defined in the `/etc/hosts` file?

toffanetto, 2024-12-16:
The lma-ubuntu1404 is in /hosts directory, but a really can't ping it. But I can ping its IP. I don't have much experience with ROS1, but I'm passing the ROS_MASTER IP, and the ros1-bridge takes the host name automatically.

Do you have any idea of how to use only the IP?

Apparently the ros-bridge sees the topics from ROS1 to ROS2 and vice-versa, but only the data can't traffic between then :(

TommyChangUMD, 2024-12-16:
hmm.  If you are already passing the ROS master's IP address (eg., `export ROS_MASTER_URI='http://192.168.1.208:11311'`), that should be good enough.  

To dig deeper, I would need to dupilcate your setup and try it myself.  Can you provide the details of your setup?

toffanetto, 2024-12-17:
Of course, thank you Tommy.

I am using a setup with two VMs in the same LAN of the host.

VM1:

- Ubuntu 22.04
- ROS2 Humble
- ROS1-Brigde of yours

VM2:

- Ubuntu 14.04
- ROS1 Indigo

TommyChangUMD, 2024-12-17:
Thanks.  That helps.  But could you also provide the VM image?   
Also the ros commands used to demonstrate the issue?

toffanetto, 2024-12-17:
Sure, I will be preparing the images for you ASAP. Thank you for your availability!

(This is the time when I think how docker makes life easier haha)

toffanetto, 2024-12-17:
Hello, @TommyChangUMD. Sorry for the delay.

I zipped and uploaded the Virtual Box VMs in this [link](https://unicampbr-my.sharepoint.com/:f:/g/personal/g289320_m_unicamp_br/EspczzMx5jREhHh5PzLwNdYBxhJisIgZfAPZ6uAgHRh4yA?e=gMiodh). Please check the VMs network adapter to bridge with your physical adapter to the two VMs stay in the same local network.

This is the procedure:

1. VM Ubuntu 14.04 
Shell 1
```
source /opt/ros/indigo/setup.bash
roscore
``` 

Shell 2
```
source /opt/ros/indigo/setup.bash
rosrun rospy_tutorials talker
``` 

2. VM Ubuntu 22.02
Shell 1
```
source /opt/ros/humble/setup.bash
source ~/ros-humble-ros1-bridge/install/local_setup.bash
export ROS_MASTER_URI='http://<VM_1_IP:11311'
ros2 run ros1_bridge dynamic_bridge --bridge-all-topics --show-introspection
```
Shell 2
```
source /opt/ros/humble/setup.bash
ros2 run demo_nodes_py listener
```

Thank you very much!

TommyChangUMD, 2024-12-18:
Hi Gabriel,

Thank you for providing the VM and the detailed instructions!
I’ll try to get to it soon, but I probably won’t have a chance until early next week or so.

I’ll follow up once I’ve had the opportunity to work on it.

TommyChangUMD, 2025-01-02:
Gabriel,

I finally had the chance to try out the VMs and instructions you provided, and they worked perfectly for me! Both the listener and talker are receiving data, and I can also see the rosbridge console output shown below.   The only thing I had to modify was the VMs' keyboard layout (to English (US).)  My guess is that your network router may be blocking something?

```
[...]
  ROS 2: /parameter_events (rcl_interfaces/msg/ParameterEvent) [1 pubs, 1 subs]
  ROS 2: /rosout (rcl_interfaces/msg/Log) [1 pubs, 0 subs]

  ROS 1: /chatter (std_msgs/String) [>0 pubs, 0 subs]
  ROS 1: /rosout (rosgraph_msgs/Log) [>0 pubs, >0 subs]
  ROS 1: /rosout_agg (rosgraph_msgs/Log) [>0 pubs, 0 subs]

created 1to2 bridge for topic '/chatter' with ROS 1 type 'std_msgs/String' and ROS 2 type 'std_msgs/msg/String'
  ROS 2: /chatter (std_msgs/msg/String) [0 pubs, 0 subs]
  ROS 2: /parameter_events (rcl_interfaces/msg/ParameterEvent) [1 pubs, 1 subs]
  ROS 2: /rosout (rcl_interfaces/msg/Log) [1 pubs, 0 subs]

[INFO] [1735853956.276127953] [ros_bridge]: Passing message from ROS 1 std_msgs/String to ROS 2 std_msgs/msg/String (showing msg only once per type)
  ROS 1: /chatter (std_msgs/String) [>0 pubs, 0 subs]
  ROS 1: /rosout (rosgraph_msgs/Log) [>0 pubs, >0 subs]
  ROS 1: /rosout_agg (rosgraph_msgs/Log) [>0 pubs, 0 subs]

  ROS 2: /chatter (std_msgs/msg/String) [0 pubs, 0 subs]
  ROS 2: /parameter_events (rcl_interfaces/msg/ParameterEvent) [1 pubs, 1 subs]
  ROS 2: /rosout (rcl_interfaces/msg/Log) [1 pubs, 0 subs]
[...]
```

toffanetto, 2025-01-03:
Hello Tommy,

Thank you very much for all your help and availability. 

I tried disconnecting my laptop from every network and running the VMs only on the local network, and after some adjustments in hostnames, it worked! I also tested my lab ROS1 package, which I will use with your bridge, and everything seems to work fine.

Again, I appreciate your help and am glad to use your repo for my master's. Please let me know if I could help with the ros1-bridge project in any way. 

Leastwise, we know that the bridge works with ROS2-Humble <> ROS1-Indigo!

An excellent year for you, Tommy!!!!

closed 2025-01-03 by toffanetto