TommyChangUMD/ros-humble-ros1-bridge-builder #23: All files being deleted when Tilix is closed

ShameerMasroor, 2025-01-02:
Hello!

I was able to run the listener and talker nodes on the two ROS versions. However, since I require particular packages to run on ROS1, I tried downloading some repos to the Tilix window that opens having ROS1. However, when I close the window and relaunch Tilix using the same command as listed in your readme, 

```bash
  rocker --x11 --user --privileged \
         --volume /dev/shm /dev/shm --network=host -- ros:noetic-ros-base-focal \
         'bash -c "sudo apt update; sudo apt install -y ros-noetic-rospy-tutorials tilix; tilix"'

```

I first downloads tilix and all (that's what I think its doing), and then launches Tilix with no older files present. How can I preserve what I downloaded earlier?

Thank you for taking out time for a beginner.

TommyChangUMD, 2025-01-02:
You can easily create a snapshot image to save and perserve what you downloaded ealier.  To do that, you use the `docker commit` command:

`$ docker commit <CONTAINER_ID> <NEW_SNAPSHOT_NAME>`

You can get the value of <CONTAINER_ID>  by running the `docker ps` command in another terminal.  In the example below, you can see that my container id is: 5753e7a2a326.

```
$ docker ps
CONTAINER ID   IMAGE          COMMAND                  CREATED              STATUS              PORTS     NAMES
5753e7a2a326   e8a242cf7933   "/ros_entrypoint.sh …"   About a minute ago   Up About a minute             stupefied_fermat
```

Once you have the container ID, you can proceed to take a snapshot of the running container and save it as a new docker image.  Like below:

`$ docker commit 5753e7a2a326 my-ros1-snapshot`

Now, the next time you want to run your snapshot with your modification preserved, just do:

`$ rocker --x11 --user --privileged --volume /dev/shm /dev/shm --network=host -- my-ros1-snapshot tilix`

TommyChangUMD, 2025-01-14:
Shameer,

If you have more questions, feel free to open a new issue.

closed 2025-01-14 by TommyChangUMD