TommyChangUMD/ros-humble-ros1-bridge-builder #24: Have you tried doing a similar thing with ros2-jazzy and noetic using ubuntu 24.04 noble?

vik748, 2025-01-30:
Hi,
We have an extensive system using ROS1 noetic and are considering using ros1-bridge to incrementally move to ROS2. This repo is the first hope that we might be able to do what we need.  Given the current ROS2 LTS version is jazzy, we are wondering what it would take to make this work with jazzy.

TommyChangUMD, 2025-02-03:
Hi Vikrant,

I will try it on Ubuntu 24.04 Jazzy and get back to you later this week.

SofiaFoguet, 2025-02-17:
Hi, did you try it? I am working on a project for a university subject. I am using Ubuntu 24.04, so I don't know how to continue the project without it. Please, could you let me know if you got it?

TommyChangUMD, 2025-02-17:
I didn’t get to it last week, but I’ll try this week.
By the way, if you could also try adapting the Dockerfile to run in Ubuntu 24.04 and let me know what errors you encounter, that would help get us both started. :)

vik748, 2025-02-17:
The main issue I imagine is the 'RUN apt-get -y install ros-desktop-dev' step. I don't think Ubuntu 24.0 has this package.

TommyChangUMD, 2025-02-18:
Here's the initial attempt.  The hard part is getting ros1 bridge to compile.  Currently it fails at 88%.  

https://github.com/TommyChangUMD/ros-humble-ros1-bridge-builder/blob/ros-jazzy-ros1-bridge-builder/Dockerfile

TommyChangUMD, 2025-03-08:
I have made more progress.  ros1 bridge now compiles and works in ros2-jazzy.   You can give it a try from the latest `ros-jazzy-ros1-bridge-builder` branch:

https://github.com/TommyChangUMD/ros-humble-ros1-bridge-builder/blob/ros-jazzy-ros1-bridge-builder/Dockerfile

Note: Looks like the `ppa:ros-for-jammy/noble` repo does not yet support ARM architecture.

TommyChangUMD, 2025-03-13:
I have created a new repo for ROS2 Jazzy + Ubuntu 24.04.   You can open new issues there.

https://github.com/TommyChangUMD/ros-jazzy-ros1-bridge-builder

closed 2025-03-13 by TommyChangUMD