TommyChangUMD/ros-humble-ros1-bridge-builder #26: Connection appears established, no access to topics or services on Humble side

lcbw, 2025-04-22:
Hello! Thank you for sharing this awesome work. 
I'm running this on Humble with Ubuntu 22.04, and followed the basic installation instructions you included. 

I have ROS1 Noetic running on another machine on my network and have confirmed I can communicate with it via the 
``` nc -v -z XXX.XXX.XXX.XXX 11311``` command. 

On my Humble machine I have the following settings: 
``` ROS_MASTER_URI='http:// XXX.XXX.XXX.XXX :11311/' ```
I've added the noetic pc to the /etc/hosts: ```  XXX.XXX.XXX.XXX noetic_pc  ```

And running: ``` ros2 run ros1_bridge dynamic_bridge --bridge-all-topics ``` spews a promising list of bridged topics and services to screen: 

```created 1to2 bridge for topic '/joint_states' with ROS 1 type 'sensor_msgs/JointState' and ROS 2 type 'sensor_msgs/msg/JointState' ```
``` Passing message from ROS 1 sensor_msgs/JointState to ROS 2 sensor_msgs/msg/JointState (showing msg only once per type) ```

But I see nothing from a terminal with the following setup: 
``` source /opt/ros/humble/setup.bash
source src/ros-humble-ros1-bridge-builder/install/local_setup.bash 
ros2 topic echo /joint_states sensor_msgs/msg/JointState
 ```
Did I forget something in my setup? Thank you for your support.

TommyChangUMD, 2025-04-22:
Is ROS1 publishing to the topic?

lcbw, 2025-04-23:
> Is ROS1 publishing to the topic?

Yes it is

TommyChangUMD, 2025-04-23:
Could you provide step-by-step instructions so that I can reproduce the issue on my end?  For example, how is the ROS1 topic being published?  What about other topics -- do they work?  Does any topic work at all, or is it just the "/joint_states" (sensor_msgs/msg/JointState) topic that isn't working?

closed 2025-04-23 by lcbw